OpenMPDK/KVSSD #2: [thread safety issue] seg fault on multithread using sync api with kvs emulator

celeryfake, 2018-09-06:
I encountered some seg faults when using kvs emulator sync api. Here is a small test program regenerate the problem.
code as follows:
`#include <stdio.h>
#include <stdint.h>
#include <mutex>
#include <atomic>
#include <unistd.h>
#include <queue>
#include <kvs_api.h>
#include <thread>

void usage(char *program)
{
  printf("==============\n");
  printf("usage: %s -d device_path [-n num_ios] [-k klen] [-v vlen]\n", program);
  printf("-d      device_path  :  kvssd device path. e.g. emul: /dev/kvemul; kdd: /dev/nvme0n1; udd: 0000:06:00.0\n");
  printf("-n      num_ios      :  total number of ios per thread\n");
  printf("-t      num_threads  :  total number of client threads\n");
  printf("-k      klen         :  key length\n");
  printf("-v      vlen         :  value length\n");
  printf("==============\n");
}

void perform_insertion(kvs_container_handle cont_hd, int offset, int count, uint16_t klen, uint32_t vlen) {

  char *key   = (char*)kvs_malloc(klen, 4096);
  char *value = (char*)kvs_malloc(vlen, 4096);
  if(key == NULL || value == NULL) {
    fprintf(stderr, "failed to allocate\n");
    exit(1);
  }

  fprintf(stdout, "\n===========\n");
  fprintf(stdout, "   Do Write Operation \n");
  fprintf(stdout, "===========\n");

  struct timespec t1, t2;
  clock_gettime(CLOCK_REALTIME, &t1);
  
  for(int i = offset; i < count + offset; i++) {
    sprintf(key, "%0*d", klen - 1, i);
    sprintf(value, "%0*d", klen - 1, i + 10); 
    
    const kvs_store_context put_ctx = { KVS_STORE_POST | KVS_SYNC_IO, 0, 0 , 0};
    const kvs_key  kvskey = { key, klen};
    const kvs_value kvsvalue = { value, vlen, 0};

    int ret = kvs_store_tuple(cont_hd, &kvskey, &kvsvalue, &put_ctx);    
    if(ret != KVS_SUCCESS) {
      fprintf(stderr, "store tuple failed with error 0x%x - %s\n", ret, kvs_errstr(ret));
      exit(1);
    } else {
      fprintf(stderr, "store key %s with value %s done %d \n", key, value, vlen);
    }
  }

  clock_gettime(CLOCK_REALTIME, &t2);
  unsigned long long start, end;
  start = t1.tv_sec * 1000000000L + t1.tv_nsec;
  end = t2.tv_sec * 1000000000L + t2.tv_nsec;
  double sec = (double)(end - start) / 1000000000L;
  fprintf(stdout, "Total time %.2f sec; Throughput %.2f ops/sec\n", sec, (double)count /sec );
  
  if(key) kvs_free(key);
  if(value) kvs_free(value);

}

int main(int argc, char *argv[]) {
  char* dev_path = NULL;
  int num_ios = 10;
  uint16_t klen = 16;
  uint32_t vlen = 4096;
  int num_thread = 1;
  int c;
 
  while ((c = getopt(argc, argv, "d:n:t:k:v:h")) != -1) {
    switch(c) {
    case 'd':
      dev_path = optarg;
      break;
    case 'n':
      num_ios = atoi(optarg);
      break;
    case 't':
      num_thread = atoi(optarg);
      break;
    case 'k':
      klen = atoi(optarg);
      break;
    case 'v':
      vlen = atoi(optarg);
      break;
    case 'h':
      usage(argv[0]);
      exit(1);
      break;
    default:
      usage(argv[0]);
      exit(0);
    }
  }

  if(dev_path == NULL) {
    fprintf(stderr, "Please specify KV SSD device path\n");
    usage(argv[0]);
    exit(0);
  }

  kvs_init_options options;
  kvs_init_env_opts(&options);
  
  options.memory.use_dpdk = 0;
  options.aio.iocomplete_fn = NULL;
  
  const char *configfile = "../kvssd_emul.conf";
  options.emul_config_file =  configfile;

  // initialize the environment
  kvs_init_env(&options);

  kvs_device_handle dev;
  kvs_open_device(dev_path, &dev);

  kvs_container_context ctx;
  kvs_create_container(dev, "test", 4, &ctx);

  kvs_container_handle cont_handle;
  kvs_open_container(dev, "test", &cont_handle);
    
  std::thread *worker = new std::thread[num_thread];

  for (int i = 0; i < num_thread; i++) {
    worker[i] = std::thread(perform_insertion, cont_handle, num_ios*i, num_ios, klen, vlen);
  }

  for (int i = 0; i < num_thread; i++) {
    worker[i].join();
  }
  kvs_close_container(cont_handle);
  kvs_delete_container(dev, "test");
  kvs_exit_env();
  
  return 0;
}`

compile script:
`~/KVSSD/PDK/core/build$ g++ test_sync.cpp -I../include  -L./ -lpthread -lkvapi -std=c++11 -fpic -Wall -g -O0`

run script:
`./a.out -d /dev/kvemul -n 100 -k 16 -v 1024 -t 4`

I haven't check the emulator source code yet. Is this a known issue?

Thanks,
Mian

changhochoi, 2018-09-07:
It will be fixed in the next release. thanks!

kvssd-support, 2019-01-28:
Since its fixed, closing this issue

closed 2019-01-28 by kvssd-support