OpenMPDK/KVSSD #6: Installation Error

mkbahk, 2019-01-04:
I followed the procedure of KVSSD_QUICK_START_GUIDE_0_7_0.pdf.
I got this kind of error message:

root@bitcoin:~/KVSSD-master/PDK/driver/PCIe/kernel_driver/kernel_v4.15# ./re_insmod.sh 
insmod: ERROR: could not insert module nvme-core.ko: Required key not available
insmod: ERROR: could not insert module nvme.ko: Required key not available

root@bitcoin:~/KVSSD-master/PDK/driver/PCIe/kernel_driver/kernel_v4.15# uname -a
Linux bitcoin 4.15.0-22-generic #24~16.04.1-Ubuntu SMP Fri May 18 09:46:31 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

Any idea for this error to fix?

kvssd-support, 2019-01-04:
This looks to be general problem in the system, not specific to KVSSD package.
Try disable Secure Boot in UEFI (BIOS) settings.

kvssd-support, 2019-01-08:
Closing this issue for now.

closed 2019-01-08 by kvssd-support