OpenMPDK/KVSSD #8: [KVCeph] ./bin/rados bench doesn't use KV-SSD, use normal SSD.

mkbahk, 2019-01-16:
Dear contributors,

First, 
I installed the KV-SSD Kernel Driver and have run the sample test program(./sample_code_async -d /dev/nvme0n1 -n 100 -q 64 -o 1 -k 16 -v 4096). I saw the use of KV-SSD(nvme0n1) upon "iostat" command.

==================iostat========================
avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           8.09    0.00    8.78    0.03    0.00   83.10

Device:            tps    kB_read/s    kB_wrtn/s    kB_read    kB_wrtn
loop0             0.00         0.00         0.00          0          0
sda              17.40         0.00       396.40          0       3964
nvme1n1           0.00         0.00         0.00          0          0
nvme0n1       54123.30         0.00    216493.20          0    2164932
=====================sample code async=================
io_complete: op = 1, key = 000000000999999, result = KVS_SUCCESS
io_complete: op = 1, key = 000000000999989, result = KVS_SUCCESS
io_complete: op = 1, key = 000000000999995, result = KVS_SUCCESS
io_complete: op = 1, key = 000000000999996, result = KVS_SUCCESS
Total time 18.47 sec; Throughput 54135.79 ops/sec
After: Total size is 3840755982336 bytes, used is 346
root@bitcoin:~/KVSSD/PDK/core/build# 
============================end of sampe...==================

second,
after build and 
[ceph.conf.txt](https://github.com/OpenMPDK/KVSSD/files/2762299/ceph.conf.txt)

install KVCeph with Kernal driver, run the rados bench command, does not use KV-SSD. which part do I have to check.

../src/vstart.sh -d -n -x -l
./bin/rados -p rbd bench 30 write

===========result of rados bench====================
   28      16       134       118   16.8549         0           -     3.28753
   29      16       134       118   16.2737         0           -     3.28753
   30      16       134       118   15.7312         0           -     3.28753
   31      16       135       119   15.3528  0.666667     7.59606     3.32374
Total time run:         31.128823
Total writes made:      135
Write size:             4194304
Object size:            4194304
Bandwidth (MB/sec):     17.3473
Stddev Bandwidth:       15.9201
Max bandwidth (MB/sec): 64
Min bandwidth (MB/sec): 0
Average IOPS:           4
Stddev IOPS:            3
Max IOPS:               16
Min IOPS:               0
Average Latency(s):     3.68917
Stddev Latency(s):      2.33163
Max latency(s):         8.72245
Min latency(s):         0.107901
Cleaning up (deleting benchmark objects)
Removed 135 objects
Clean up completed and total clean up time :0.097754
root@bitcoin:~/KVSSD/application/KVCeph/build# 

===========result of iostat============================
 avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           2.63    0.00    2.24    8.60    0.00   86.53

Device:            tps    kB_read/s    kB_wrtn/s    kB_read    kB_wrtn
loop0             0.00         0.00         0.00          0          0
sda             203.20         0.00    132416.40          0    1324164
nvme1n1           0.00         0.00         0.00          0          0
nvme0n1           0.00         0.00         0.00          0          0
=================end of result================

regards,
Moon-Kee Bahk
m k b a h k@atto-research.com

ywkang-ssi, 2019-01-17:
It looks like CEPH is not properly configured. Each OSD requires one device. When there are multiple OSDs, 'kvsstore_dev_path' needs to be set per OSD under [osd.x] section in the configuration file.  For example, 
[osd.0]
        host = xxxx
        public addr = xxx
        cluster addr = xxx
        key = xxxx
        log file = ....../ceph-deploy/osd.0/osd.log
        osd data = ....../ceph-deploy/osd.0/data
        keyring =  ....../ceph-deploy/ceph.mon.keyring
        kvsstore_dev_path = /dev/nvme2n1 
        run dir =  ....../ceph-deploy/run

Please also check the out/osd.0.log file to make sure the KV SSD device is opened. When successful, KvsStore prints out the following message in the log file.  
    "KADI is opened successfully, fd x, efd xx, dev /dev/nvme0n1"
Since there are three OSDs, there should be three open messages with different device paths.

kvssd-support, 2019-03-05:
Hope the latest comment would have helped to resolve this issue. Closing this bug.

closed 2019-03-05 by kvssd-support