OpenMPDK/KVSSD #11: [KVBench] Running kv_bench with iterator causes error

soojin980213, 2019-01-25:
I'm testing kv_bench with kernel device driver and every time I try to run kv_bench with iterator when write_mode=async, it fails with KVS_ERR_SYS_IO error followed by segmentation fault.

[Steps]
(1) When with_iterator option is set as 'false'(no iterator), it terminates normally. However, when I set it as 'true' or 'alone', it becomes buggy, terminates with segmentation fault, and everything on my machine freezes.
(2) Even 'reboot' or 'shutdown' command doesn't work, so I should shutdown and reboot machine compulsorily.
(3) When I check if everything is fine with test program(usually sample_code_async) after reboot, it fails with KVS_ERR_DEV_INIT.
(4) I recompile/reinstall NVMe modules for KVSSD to resolve the error, but system doesn't recognize nvme controller and block device after that. To solve this problem, I reboot machine again.
(5) Now KVSSD Firmware is damaged, shown as 'ERRORMOD'. Luckily, I can recover KVSSD Firmware successfully every time it is damaged. However, when I try to run kv_bench with same settings again, same error rises again.

Below, the line marked as '=>' is where I encounter segmentation fault. 
('mov (%rdx), %r8' at assembly code)
(KVSSD/application/kvbench/wrappers/couch_kv.cc)
----------------------------------------------------------------------------------------------------------------------------------
void on_io_complete(kvs_callback_context* ioctx) {
  if(ioctx->result != KVS_SUCCESS && ioctx->result != KVS_ERR_KEY_NOT_EXIST) {
=>    fprintf(stdout, "io error: op = %d, key = %s, result = %s\n", ioctx->opcode, (char*)ioctx->key->key, kvs_errstr(ioctx->result));
exit(1);
}
----------------------------------------------------------------------------------------------------------------------------------

Until now, I couldn't find any information related with this error. Is this a known issue?

kvssd-support, 2019-01-25:
Please share the kv_bench config file along with details mentioned in bug template.

soojin980213, 2019-01-28:
1. Error message: Since everything freezes when this error occurs, I was unable to take screenshot for this. Instead, I'll write down the error message.
------------------------------------------------
OTHER failed with KVS_ERR_SYS_IO
Segmentation Fault (Core Dumped)
------------------------------------------------

2. Steps to reproduce error: Please check my first comment.

3. System environment)
 - Firmware version : EGA50K01
 - Number of SSDs : 1 (block device: nvme0n1 / controller: nvme0)
 - OS & Kernel version : Ubuntu 16.04 Kernel v4.13 
 - GCC version : v5.4.0
 - kvbench version : v0.7.0
 - KV API version : v0.7.0
 - User driver version : v0.7.0 (But I'm not using UDD)
 - Driver : KDD

4. Expected performance: kvbench terminates normally without any error.

soojin980213, 2019-01-28:
I've just found that v0.7.1 is released, so I'll test with that version if same error occurs.

kvssd-support, 2019-01-30:
We are looking into this issue - will update

kvssd-support, 2019-02-26:
This issue is fixed, will update the latest repo shortly.

soojin980213, 2019-02-26:
Now it does not terminate with segmentation fault anymore, but trial of using iterator with KDD still fails with KVS_ERR_SYS_IO.
I guess the last part(including polling / interrupt) of open_iterator() function is buggy, but I'm not sure about that.
Are there any solutions or bug reports about such problem?

kvssd-support, 2019-02-27:
Please share the detailed logs.

soojin980213, 2019-02-27:
bench_config.ini and all environments are same, except KV API version - v0.8.0 for now.

[image]
<screenshot>

kvssd-support, 2019-02-27:
Thanks for the information, we will look into this issue (as its working in our environment)

However, please confirm the following (to narrow down the problem):
a) if sample_tests (under sample_code) are working fine using Iterator (not using kv_bench)
b) Try to rebuild/recompile the kvbench and run the same test with iterator

soojin980213, 2019-02-27:
Thanks for your fast response.
However, I'm so sorry to say that the error is suddenly disappeared and I can't reproduce it anymore.
I've recompiled KV API for several times, and suddenly kv_bench started to work well with iterator, after one of my trials of recompilation.

I've done something like that from yesterday to today: Compile KDD several times -> Try to use emulator -> Compile KDD again, editing kvkdd.cpp -> kv_bench with KDD and iterator started to work -> Compile KDD, removing what I've done -> kv_bench still works
This is the screenshot of the code that I've edited in kvkdd.cpp
[image]

I still have no idea why it hadn't work and why it is working now.
Anyway, this is the additional information for narrowing the problem.
a) sample_tests always worked well, except the case of facing KVS_ERR_ITERATOR_OPEN error. (see #14)
b) I haven't tried it, because now I'm keep failing to reproduce the error.

kvssd-support, 2019-02-28:
Thanks for the detailed information.
Try to upgrade to the latest firmware, which should have fix for usage of KVSSD.

kvssd-support, 2019-03-04:
Closing this issue as the original issue is resolved.

closed 2019-03-04 by kvssd-support