OpenMPDK/KVSSD #14: [KV API] About closing iterator when test is done

soojin980213, 2019-02-26:
When testing / running benchmark for KVSSD(using KDD) with iterator is done, does it automatically closes all iterators, even when the program doesn't terminates normally - rising error(i.e. KVS_ERR_SYS_IO, etc)?

kvssd-support, 2019-02-27:
API: kvs_close_iterator_all(kvs_container_handle cont_hd), this will close all opened iterators in case an iterator is not closed properly during last run. Currently it is only supported in SPDK (udd) driver. Will add this to KDD soon .

soojin980213, 2019-02-27:
Actually, I'm facing the situation that once KVS_ERR_ITERATOR_OPEN error rises, it rises over and over again in every trial of running benchmark / testing, while using KDD.
Now, my solution for this problem is, to recompile and reinstall the kernel driver. According to your answer, it seems that there is no solution/automatic recovery using KV API functions for this situation now - am I right?
If not, are there any available solutions for this?

kvssd-support, 2019-02-27:
To resolve this issue, try following options:

a) For testing purpose, bitmask (by default set to '0xffff0000') can be changed under file 'KVSSD/application/kvbench/wrappers/couch_kv.cc" and recompile 'kv_bench' application. This should get rid of KVS_ERR_ITERATOR_OPEN error

b) Use KV API  'kvs_close_iterator_all' - this will close all opened iterators in case an iterator is not closed properly during last run. Tried adding the KVAPI under the Sample_code (core/sample_code/test_async.cpp) and attempt to close iterator by running the "sample_code" if there any already open iterator using SPDK driver.

kvssd-support, 2019-03-04:
Hope this issue can be closed as KVS_ERR_ITERATOR_OPEN error is not any more reproducible.

kvssd-support, 2019-03-05:
Closing this issue.

closed 2019-03-05 by kvssd-support