OpenMPDK/KVSSD #15: Cant build on Ubuntu

kreuzerkrieg, 2019-03-02:
for some odd reason I'm not able to build master/0.7.0/0.8.1
Tested on my dev machine and on clean AWS instance with ubuntu 18.04 with latest updates installed

> git clone https://github.com/OpenMPDK/KVSSD.git
> cd KVSSD/
> cd PDK/core/
> ./tools/install_deps.sh
> mkdir build
> cd build/
> cmake -DWITH_EMU=ON ../
> make -j

so, first I encounter compilation error 

> /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/include/private/history.hpp:139:30: error: expected ‘)’ before ‘<’ token
>      latency_model(std::vector<double> iops_model_coefficients) {
>                               ^
> In file included from /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/include/private/kv_emulator.hpp:45:0,
>                  from /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/emulator/src/kv_emulator.cpp:41:
> /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/include/private/history.hpp:277:27: error: expected ‘)’ before ‘<’ token
>      kv_history(std::vector<double> iops_model_coefficients) : model(iops_model_coefficients) {}
>                            ^
> In file included from /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/emulator/src/kv_emulator.cpp:41:0:
> /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/include/private/kv_emulator.hpp:115:41: error: ‘std::vector’ has not been declared
>      kv_emulator(uint64_t capacity, std::vector<double> iops_model_coefficients, bool_t use_iops_model, uint32_t nsid);
>                                          ^~~~~~
> /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/include/private/kv_emulator.hpp:115:47: error: expected ‘,’ or ‘...’ before ‘<’ token
>      kv_emulator(uint64_t capacity, std::vector<double> iops_model_coefficients, bool_t use_iops_model, uint32_t nsid);
>                                                ^
> In file included from /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/include/private/kv_emulator.hpp:45:0,
>                  from /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/emulator/src/kv_emulator.cpp:41:
> /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/include/private/history.hpp:139:30: error: expected ‘)’ before ‘<’ token
>      latency_model(std::vector<double> iops_model_coefficients) {
>                               ^
> In file included from /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/include/private/kv_emulator.hpp:45:0,
>                  from /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/emulator/src/kv_emulator.cpp:41:
> /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/include/private/history.hpp:277:27: error: expected ‘)’ before ‘<’ token
>      kv_history(std::vector<double> iops_model_coefficients) : model(iops_model_coefficients) {}
>                            ^
> /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/emulator/src/kv_emulator.cpp:49:50: error: ‘std::vector’ has not been declared
>  kv_emulator::kv_emulator(uint64_t capacity, std::vector<double> iops_model_coefficients, bool_t use_iops_model, uint32_t nsid): stat(iops_model_coefficients), m_capacity(capacity),m_available(capacity), m_use_iops_model(use_iops_model), m_nsid(nsid) {
>                                                   ^~~~~~
> In file included from /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/emulator/src/kv_emulator.cpp:41:0:
> /home/ubuntu/KVSSD/PDK/core/src/device_abstract_layer/include/private/kv_emulator.hpp:115:41: error: ‘std::vector’ has not been declared
>      kv_emulator(uint64_t capacity, std::vector<double> iops_model_coefficients, bool_t use_iops_model, uint32_t nsid);
>                                          ^~~~~~

I've added `vector` include to the ./src/device_abstract_layer/include/private/history.hpp and it compiled and linked

kvssd-support, 2019-03-04:
Ubuntu 18.04 is not supported configuration.

kreuzerkrieg, 2019-03-04:
any plans to support 18.04 in nearest future?

kvssd-support, 2019-03-05:
Please share the kernel version along with details mentioned in bug template.

kreuzerkrieg, 2019-03-06:
- Firmware version : N/A
 - Number of SSDs :
 - OS & Kernel version [e.g., Ubuntu 16.04 Kernel v4.9.5]:  Ubuntu 18.04 Kernel: 5.0
 - GCC version [e.g., gcc v5.0.0] : GCC 7.3
 - kvbench version if kvbench runs [e.g., v0.6.0]: N/A
 - KV API version [e.g., v0.6.0] 0.8.0
 - User driver version :
 - Driver [Kernel or user driver or emulator] : emulator (kernel in the nearest future)

kvssd-support, 2019-03-06:
Could you please try Ubuntu 18.04 with kernel v4.15 (as kernel 5.0 is not supported) ..

kreuzerkrieg, 2019-03-06:
We already tried emulator with 4.15 on Ubuntu 18.04 (after fixing the above inclusion) and it worked as expected. We also going to test it with latest stable kernel, id est 5.0 after making necessary changes.

kvssd-support, 2019-03-11:
Hope we can close this issue.

kreuzerkrieg, 2019-03-11:
It depends. You definitely can close the issue if you dont intend to make the kvssd work out of the box for Ubuntu 18.04.
On the other hand it is one-liner fix, I can create PR if you accept any.

kvssd-support, 2019-03-21:
We are looking into this - will update

xadongfei, 2019-08-28:
Your running result is correct. 
For the emulator, the device path is only used to identify different emulators, the data is stored in the memory, not persisted to the corresponding path on the disk, so you can't find in the emulator device in your system.

closed 2019-11-12 by hao86yan