OpenMPDK/KVSSD #16: Documentation claims no CentOS 7 support.

deponty, 2019-03-15:
**Location (Korea, USA, China, India, etc.)**
San Diego

**Describe the bug**
Requiring gcc v. 5.0 precludes building on CentOS 7

**To Reproduce**
1.  Try a base CentOS 7 distribution build.
2.  Please refer to the documentation.

**Expected behavior**
Unknown - I haven't tried it.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System environment (please complete the following information)**
 - OS & Kernel version [e.g., Ubuntu 16.04 Kernel v4.9.5]: 
      CentOS 7
      3.10.0-957.5.1.el7.x86_64

 - GCC version [e.g., gcc v5.0.0] :
      gcc 4.8.5 20150623 (Red Hat 4.8.5-36)

**Additional context**
david.evans@samsung.com

kvssd-support, 2019-03-18:
Please refer the document mentioned support of CenOS and required gcc verion:
https://github.com/OpenMPDK/KVSSD/blob/master/KVSSD_QUICK_START_GUIDE_0.7.1.pdf

kvssd-support, 2019-03-21:
Closing this issue, as no further comments

closed 2019-03-21 by kvssd-support