OpenMPDK/KVSSD #17: Spec 0.7.0 typo

deponty, 2019-03-15:
**Location (Korea, USA, China, India, etc.)**
San Diego

**Describe the bug**
Section 6.2.1:
typedef enum {
KVS_ITERAOR_KEY_VALUE =1,
} kvs_iterator_type; 

should be
typedef enum {
    KVS_ITERATOR_KEY_VALUE = 1,
} kvs_iterator_type; 
----------------------------------------------------

Also, this isn't clear:
// [OPTION] iterator command retrieves key and
delete

Does iterator with delete return the key and value before deleting the tuple, or does it just return the key?

**To Reproduce**
Steps to reproduce the behavior:
1.  Read the spec.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System environment (please complete the following information)**
 - Firmware version :
 - Number of SSDs :
 - OS & Kernel version [e.g., Ubuntu 16.04 Kernel v4.9.5]: 
 - GCC version [e.g., gcc v5.0.0] :
 - kvbench version if kvbench runs [e.g., v0.6.0]:
 - KV API version [e.g., v0.6.0]
 - User driver version :
 - Driver [Kernel or user driver or emulator] :

**Workload**
 - number of records or data size
 - Workload(insert, mixed workload, etc.) [e.g., sequential or random insert, or 50% Read & 50% write]
 - key size :  
 - value size : 
 - operation option if available [e.g., sync or async mode] : 

**Additional context**
Add any other context about the problem here.

kvssd-support, 2019-03-18:
We are looking into this..

kvssd-support, 2019-03-19:
Typo error will be fixed in the next release.

As per Samsung_KV_API_spec_v0.7.pdf - kvs_open_iterator, Samsung device support KVS_ITERATOR_KEY iterator type only. Currently, other two options KVS_ITERATOR_KEY_VALUE and KVS_ITERATOR_WITH_DELETE are not supported

kvssd-support, 2019-05-14:
Please use the latest code as its fixed in Spec

closed 2019-05-14 by kvssd-support