OpenMPDK/KVSSD #18: Spec 0.7 does not match kvs_api.h

deponty, 2019-03-18:
**Location (Korea, USA, China, India, etc.)**
USA

**Describe the bug**
A clear and concise description of what the bug is.

In kvs_api.h:
  #define KVS_ALIGNMENT_UNIT 4

In the KV SSD API spec:
  6.1.1 KVS_ALIGNMENT_UNIT
  This is an alignment unit. An offset of value must be a multiple of this value.
  [SAMSUNG] The default alignment unit for the Samsung key-value SSD is 32 bytes.

**To Reproduce**
Steps to reproduce the behavior:
1.  
2. 
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System environment (please complete the following information)**
 - Firmware version :
 - Number of SSDs :
 - OS & Kernel version [e.g., Ubuntu 16.04 Kernel v4.9.5]: 
 - GCC version [e.g., gcc v5.0.0] :
 - kvbench version if kvbench runs [e.g., v0.6.0]:
 - KV API version [e.g., v0.6.0]
 - User driver version :
 - Driver [Kernel or user driver or emulator] :

**Workload**
 - number of records or data size
 - Workload(insert, mixed workload, etc.) [e.g., sequential or random insert, or 50% Read & 50% write]
 - key size :  
 - value size : 
 - operation option if available [e.g., sync or async mode] : 

**Additional context**
Add any other context about the problem here.

kvssd-support, 2019-03-21:
We will update the Spec as needed in the next release

kvssd-support, 2019-05-14:
Please use the latest repo as part of fixed release.

closed 2019-05-14 by kvssd-support