OpenMPDK/KVSSD #19: kvs_init_* not documented

deponty, 2019-03-18:
**Location (Korea, USA, China, India, etc.)**
USA

**Describe the bug**
A clear and concise description of what the bug is.

The KV_ADI spec does not document anything related to initialization functions, though they are clearly used in the sample code and in the kvs_api.h

The comments in kvs_api.h don't tell me anything about what the impact of these initialization variables in kv_init_options do.

**To Reproduce**
Steps to reproduce the behavior:
1.  
2. 
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System environment (please complete the following information)**
 - Firmware version :
 - Number of SSDs :
 - OS & Kernel version [e.g., Ubuntu 16.04 Kernel v4.9.5]: 
 - GCC version [e.g., gcc v5.0.0] :
 - kvbench version if kvbench runs [e.g., v0.6.0]:
 - KV API version [e.g., v0.6.0]
 - User driver version :
 - Driver [Kernel or user driver or emulator] :

**Workload**
 - number of records or data size
 - Workload(insert, mixed workload, etc.) [e.g., sequential or random insert, or 50% Read & 50% write]
 - key size :  
 - value size : 
 - operation option if available [e.g., sync or async mode] : 

**Additional context**
Add any other context about the problem here.

kvssd-support, 2019-03-21:
We will update the Spec as needed in the next release

kvssd-support, 2019-05-14:
Please use the latest repo for fix

closed 2019-05-14 by kvssd-support