OpenMPDK/KVSSD #21: Repo is too big to clone

deponty, 2019-03-26:
**Location (Korea, USA, China, India, etc.)**
Put your location for prompt support.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1.  
2. 
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System environment (please complete the following information)**
 - Firmware version :
 - Number of SSDs :
 - OS & Kernel version [e.g., Ubuntu 16.04 Kernel v4.9.5]: 
 - GCC version [e.g., gcc v5.0.0] :
 - kvbench version if kvbench runs [e.g., v0.6.0]:
 - KV API version [e.g., v0.6.0]
 - User driver version :
 - Driver [Kernel or user driver or emulator] :

**Workload**
 - number of records or data size
 - Workload(insert, mixed workload, etc.) [e.g., sequential or random insert, or 50% Read & 50% write]
 - key size :  
 - value size : 
 - operation option if available [e.g., sync or async mode] : 

**Additional context**
Add any other context about the problem here.

deponty, 2019-03-26:
Our lab machines are really space limited. Ceph and binaries are causing my machine to run out of space.

kvssd-support, 2019-04-01:
KVSSD git repo restructuring will happen soon

deponty, 2019-04-01:
Okay, thanks. It's generally considered bad practice to check in compiled binaries for space reasons - though once they're out in the wild eliminating them may be very difficult. Something for you to consider.

kvssd-support, 2019-05-14:
Please use the latest repo as KVSSD repo is segregated

closed 2019-05-14 by kvssd-support