OpenMPDK/KVSSD #22: queue full errors result in io_cmd leaks

deponty, 2019-03-26:
**Location (Korea, USA, China, India, etc.)**
USA 

**Describe the bug**
In function kv_store in kv_device.cpp, an io_cmd object is allocated and passed to submit_io. If there are any errors returned, the kv_store function goes out of scope and leaks the io_cmd object.

There are several of these types of bugs in the code (I count at least a dozen).

**To Reproduce**
Steps to reproduce the behavior:
1.  Use the kernel driver to overflow a queue. 
2.  Once it hits a QFULL the code leaks memory.
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System environment (please complete the following information)**
 - Firmware version :
 - Number of SSDs :
 - OS & Kernel version [e.g., Ubuntu 16.04 Kernel v4.9.5]: 
 - GCC version [e.g., gcc v5.0.0] :
 - kvbench version if kvbench runs [e.g., v0.6.0]:
 - KV API version [e.g., v0.6.0]
 - User driver version :
 - Driver [Kernel or user driver or emulator] :

**Workload**
 - number of records or data size
 - Workload(insert, mixed workload, etc.) [e.g., sequential or random insert, or 50% Read & 50% write]
 - key size :  
 - value size : 
 - operation option if available [e.g., sync or async mode] : 

**Additional context**
Add any other context about the problem here.

kvssd-support, 2019-03-27:
Please share the details mentioned in bug template and steps to reproduce along with logs.

kvssd-support, 2019-04-05:
Next release will fix this issue

closed 2020-03-23 by hao86yan