OpenMPDK/KVSSD #23: KDDriver::store_tuple() leaks memory in default case.

deponty, 2019-03-26:
**Location (Korea, USA, China, India, etc.)**
San Diego, USA (david.evans@samsung.com)

**Describe the bug**
KDDriver::store_tuple() leaks memory because prep_io_context allocates memory that doesn't get freed if the default case gets hit.

**To Reproduce**
Steps to reproduce the behavior:
1.  
2. 
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System environment (please complete the following information)**
 - Firmware version :
 - Number of SSDs :
 - OS & Kernel version [e.g., Ubuntu 16.04 Kernel v4.9.5]: 
 - GCC version [e.g., gcc v5.0.0] :
 - kvbench version if kvbench runs [e.g., v0.6.0]:
 - KV API version [e.g., v0.6.0]
 - User driver version :
 - Driver [Kernel or user driver or emulator] :

**Workload**
 - number of records or data size
 - Workload(insert, mixed workload, etc.) [e.g., sequential or random insert, or 50% Read & 50% write]
 - key size :  
 - value size : 
 - operation option if available [e.g., sync or async mode] : 

**Additional context**
Add any other context about the problem here.

kvssd-support, 2019-03-27:
Please share the details mentioned in bug template and steps to reproduce along with logs.

deponty, 2019-03-27:
The kernel driver version is 3.10.

kvssd-support, 2019-04-05:
Next release will fix this issue

closed 2020-03-23 by hao86yan