OpenMPDK/KVSSD #26: [KV API] Some questions about iterator

terrafin, 2019-04-11:
Dear all,
I have some questions about iterator's functionalities.

1. Does KVSSD Host SW Packages supports KVS_ITERATOR_KEY_VALUE mode now(for version 0.9.0)? According to Samsung_KV_API_spec, it doesn't. However, I saw some lines that once used for restricting KVS_ITERATOR_KEY_VALUE mode but now commented out, in source codes.

2. What does kvs_iterator_next() function do exactly? Does it concatenate data(key, key length, etc) for only one key to it_list?

Thank you.

kvssd-support, 2019-04-15:
The Samsung Key-value SSD currently supports only KVS_ITERATOR_KEY option only. 
When KVS_ITERATOR_KEY is set, multiple keys are retrieved for every kvs_iterator_next() API call - (a subset of keys are returned in iter_list->it_list data structure)

terrafin, 2019-04-18:
Then, what is the rule of retrieving keys when kvs_iterator_next() is called once? A group? Or, just bitmask and pattern?

kvssd-support, 2019-04-24:
Please refer Samsung KV API spec, section 7.4.5 "kvs_iterator_next" for more details on how keys are retrieved when kvs_iterator_next() is called. 

In-case still you have any query, please share query.

kvssd-support, 2019-05-06:
Closing this issue now

closed 2019-05-06 by kvssd-support