OpenMPDK/KVSSD #27: [KVSSD_Firmware] firmware update problem

joonhyuk96sj, 2019-04-16:
**Location (Korea, USA, China, India, etc.)**
Korea

**Describe the bug**
Can't perform firmware update via nvme-cli 'nvme-fw-activate' command.

**To Reproduce**
Steps to reproduce the behavior:
1. After firmware was damaged(ERRORMOD), tried to do update by using new firmware.
   Followed instructions from "KVSSD_seminar_2018_fw_introduction.pdf".
2.  Install nvme-cli package
3.  Do "nvme fw-download" -> Firmware download success
4.  Do "sleep 10"
5.  Do "nvme fw-activate"     -> NVME Admin command error:FIRMWARE_IMAGE(2107)

**Expected behavior**
After step5, i expected a log "Success activating action:1, slot:2, but a conventional reset is required" 
(According to "KVSSD_seminar_2018_fw_introduction.pdf", use slot 2 & action 1)

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**System environment (please complete the following information)**
 - Firmware version : EEA51K0Q_20181221_ENC.bin(NEW), EGA50K01(ERRORMOD)
 - Number of SSDs  : SAMSUNG MZQLB3T8HALS-000KV(Model), S4DRNY0KA00033(SN) 
 - OS & Kernel version :  Ubuntu 16.04, Kernel 4.4.0-131-generic
 - GCC version :   v5.4.0 
 - kvbench version if kvbench runs [e.g., v0.6.0]:
 - KV API version [e.g., v0.6.0]
 - User driver version :
 - Driver [Kernel or user driver or emulator] : 

**Workload**
 - number of records or data size
 - Workload(insert, mixed workload, etc.)
 - key size :  
 - value size : 
 - operation option if available [e.g., sync or async mode] : 

**Additional context**

terrafin, 2019-04-18:
To me, following these steps worked:
 $ sudo nvme fw-download /dev/nvme0 --fw=<Binary name> --xfer=0x20000
 $ sudo nvme format /dev/nvme0
 $ sudo nvme fw-activate /dev/nvme0 --slot=0 --action=1
 $ sudo nvme reset /dev/nvme0

joonhyuk96sj, 2019-04-22:
It works, thank you!

closed 2019-04-22 by joonhyuk96sj

CS0522, 2025-05-22:
Hello, may I ask where I should find the firmware binary files like 'EEA51K0Q_20181221_ENC.bin'?

CS0522, 2025-05-22:
And if I have a normal Samsung NVMe SSD, I can flush this firmware to make it as a KV-SSD?