OpenMPDK/KVSSD #28: [Question] what makes the maximum key/value size to 255B/2MB, respectively

Jhris, 2019-05-01:
Dear all,

I have a tiny question about KVSSD system's features.

I have noticed that 
1) KVSSD use KV interface which is extended from NVMe standard
2) the maximum key value size of KVSSD is 255B, 2MB, respectively

but I can't get about 'what makes the maximum key/value size to 255B/2MB, respectively'.
(simply, it can be considered that maximum value size can be up to 512MB, because 4 byte is allocated for 'value size' in KV IF)

Is there any specific reason for these limitations?

kvssd-support, 2019-05-02:
Currently, KV SSD firmware makes the limitation to support KVS maximum key/value size to 255B/2MB, respectively. However, in the future version firmware, larger key-value length might be supported.

Jhris, 2019-05-02:
To. kvssd-support.

first of all, thank you for your reply.

but, as i wrote, i have already noticed the limitation of maximum key/value size.

i just want to get some information about the reason for this limitation.

which characteristics of the storage device make the key-value length limited?
or Is that a firmware-side issue, not the device-side issue?

kvssd-support, 2019-05-06:
This is supported due to initial requirements and SSD hardware resource management complexity. As mentioned, in the future version firmware, larger key-value length might be supported.

closed 2019-05-06 by kvssd-support