OpenMPDK/KVSSD #29: [kvbench] About KVS_ERR_DEV_CAPACITY error

terrafin, 2019-05-02:
I tried to run kvbench(KVS) with 5 billion documents and 5 billion operations, but it stopped with KVS_ERR_DEV_CAPACITY Error during benchmark period.
However, when I checked the current state of KVSSD with 'sudo nvme list' command, it turned out that only 5.12MB is used.
Is there any internal restriction about usable capacity of KVSSD, for single container?

kvssd-support, 2019-05-02:
Please share the complete system / KVSSD logs (mentioned in bug format) along with the details of the documents size of each documents of 5 billion.

terrafin, 2019-05-02:
[bench_config.txt](https://github.com/OpenMPDK/KVSSD/files/3137815/bench_config.txt)
[image]

System configuration)
    Firmware version : EEA51K0Q
    Number of SSDs : 1 (block device: nvme0n1 / controller: nvme0)
    OS & Kernel version : Ubuntu 16.04 Kernel v4.13
    GCC version : v5.4.0
    kvbench version : v0.8.0 (Latest release)
    KV API version : v0.9.0 (Latest release)
    Driver : KDD

hao86yan, 2019-05-06:
KV SSD re-defined the field "Name Utilization (NUSE)" in NVMe Identify Namespace Data Structure. In standard NVMe specification, NUSE indicates the current number of logical blocks allocated in the namespace. For KV SSD, this field indicates the current ratio of capacity allocated in namespace. Its value ranges from 0 (0.00%) to 10000 (100.00%). So, 5.12MB of nvmecli represents the whole capacity of KV SSD has been used. (5.12MB * 1000 * 1000) / 512 = 10000.

terrafin, 2019-05-07:
Then, in order to check real capacity, should I use nvme_id_ns::nuse and nvme_id_ns::nsze?

hao86yan, 2019-05-07:
Yes. At present, nvme_id_ns::nsze indicates the total capacity of KV SSD and nvme_id_ns::nsze indicates the current ratio of capacity that has been used. If you want to get the free capacity, you can calculate it through following formular:
free capacity = nvme_id_ns::nsze * (1.0 - nvme_id_ns::nuse / 10000.0)

terrafin, 2019-05-07:
Okay. Thanks a lot!

terrafin, 2019-05-28:
@hao86yan 
Is there any future plan for increasing the possible range of nvme_id_ns::nuse value, for more precised calculation of used capacity?

hao86yan, 2019-05-29:
@terrafin 
This feature is implemented by the SSD firmware. So far, we didn't get any development plan about this feature from the firmware development team. But I could propose this requirement to firmware development team and keep you updated if there is any progress.

terrafin, 2019-05-31:
@hao86yan 
Alright. Thank you!

closed 2019-05-31 by terrafin