OpenMPDK/KVSSD #32: Regarding iterator and get_capacity

tqtrunghnvn, 2019-06-24:
**Location (Korea, USA, China, India, etc.)**
Korean.

**Describe the bug**
When deleting the tuple using kvs_delete_tuple_async(), the capacity of KVSSD has no change. Moreover, the capacity received by API is different with 'sudo nvme list' command. 

**To Reproduce**
Steps to reproduce the behavior:
1. Use iterator (bitmask=0x80000000, bit_pattern=0x8000000) to get keys (I already wrote too many key-value pairs to KVSSD, they are about 250GB).
2. In print_iterator_keyvals() function, I checked if the key-value pair exists, then deleting them using kvs_delete_tuple_async().
3. I saw that there are too many key-value pairs were deleted.

**Expected behavior**
1. After I delete key-value pairs using kvs_delete_tuple_async() function, the capacity of KVSSD should be decreased but it has no change.
2. Moreover, after I had deleted key-value pairs, I tried to use iterator again and it still retrieves keys which I used to delete values as above --> it wastes time when I use iterator again
3. The capacity of KVSSD checked by Samsung's APIs is different with 'sudo nvme list' command.

**Screenshots**
Capacity of KVSSD checked by APIs
[image]

Capacity of KVSSD checked by 'sudo nvme list'
[image]

**System environment (please complete the following information)**
 - Firmware version : EEA51K0Q
 - Number of SSDs : 1 (block device: nvme0n1 / controller: nvme0)
 - OS & Kernel version: Ubuntu 16.04 Kernel v4.4.0-131-generic
 - GCC version: 5.4.0 20160609 
 - KV API version: 1.0.0 (latest) 
 - Driver: KDD (kernel_v4.4.0-141-ubuntu-16_04)

hao86yan, 2019-06-25:
Hi tqtrunghnvn,

Samung KV SSD redefined the Namespace Utilization field of NVMe spec. So, the nvme tool couldn't get the right capacity utilization of KV SSD.  Now, Samsung already developed new firmware to keep the namespace utilization field of KV SSD consistent with standard NVMe spec and we will fix this issue in next release. 

About your deleting issue, I recommend you upgrade SSD to newer firmware.

tqtrunghnvn, 2019-06-25:
Thank you for your response,

Could you please tell me where I can get the latest firmware?

hao86yan, 2019-06-27:
At present, there is no way to get the firmware at outside of Samsung.  
You could try to get the latest firmware by the same way you got your current KV firmware. 
Now, we are planning to upload the firmware to the GibHub.  After after, you could get the firmware from the GitHub.

tqtrunghnvn, 2019-07-01:
Thank you for your answer!

closed 2019-07-01 by tqtrunghnvn