OpenMPDK/KVSSD #35: [KVS] Latest version of CMakeLists.txt for kvbench is likely buggy

terrafin, 2019-07-29:
**Location (Korea, USA, China, India, etc.)**
Korea

**Describe the bug**
Compiling kvbench with KVSSD Host SW Packages of latest version(v1.2.0) fails.
KVSSD/application/kvbench/CMakeLists.txt seems to be a reason of such problem, since changing CMakeLists.txt results on successful compilation.

To prevent an compilation error, I've changed CMakeLists.txt like this:
**160 #set(KVS_LIB -L${CMAKE_LIBRARY_PATH} -lkvapi)
161 target_link_libraries(kv_bench ${COMMON_LIB} ${CMAKE_LIBRARY_PATH})
=>
160 set(KVS_LIB -L${CMAKE_LIBRARY_PATH} -lkvapi)
161 target_link_libraries(kv_bench ${COMMON_LIB} ${KVS_LIB})**
, as it was in older version.

**Screenshots**
[image]

**System environment**
 - Firmware version : EEA51K0Q
 - Number of SSDs : 1 (nvme0n1)
 - OS & Kernel version: Ubuntu 18.04 Kernel v4.15.0
 - GCC version: gcc v7.4.0
 - kvbench version: v1.2.0 (Iatest)
 - KV API version: v1.2.0 (latest)
 - User driver version : uNVMe 18.11
 - Driver: Both KDD and UDD result on same error

hao86yan, 2019-07-30:
This is caused by the incorrect library path. To fix the CMake warnning issue, the cmake was  updated. Now, you need to compile the kv_bench through:
cmake -DCMAKE_INCLUDE_PATH=/KVSSD/PDK/core/include -DCMAKE_LIBRARY_PATH=/KVSSD/PDK/core/build/libkvapi.so ../

You can get the compilation guide  inthe readme of kv_bench (KVSSD / application / kvbench / README )

terrafin, 2019-07-30:
@hao86yan Okay. Thank you so much!

closed 2019-07-30 by terrafin