OpenMPDK/KVSSD #36: [kvbench] Running kvbench with UDD in synchronous mode keep fails.

terrafin, 2019-08-05:
In order to use KVSSD with UDD in synchronous mode, I've done setup during device initialization, by setting 2nd argument of KUDDriver::init() as 'true', at line 480 of cfrontend.cpp.
Since sample_code_sync terminated normally and sample_code_async terminated with error KVS_ERR_DD_INVALID_QUEUE_TYPE, I guessed that setup was successful.

However, when I tried to run kvbench with UDD, it only worked with asynchronous mode(in other words, only when when write_mode = async at [kvs] section).
And every time I tried to run it in synchronous mode, uNVMe driver function terminated with  KVS_ERR_DD_INVALID_QUEUE_TYPE.

Am I doing something wrong? How can I run kv_bench with UDD, in synchronous mode?

jackcost6, 2019-08-05:
Hi terrafin,
I have reproduced this issue. This belongs to the issue of uNVMe driver. We will release uNVMe driver recently.
If you want to run kv_bench with UDD in synchronous mode now, you can modify write_mode=async at kvs section in configuration file and bind run thread to core 0.
The method of binding run thread to core 0 is as following:
1. modify core_ids at [kvs] section in bench configuration file. 
      core_ids = 1,3,5  ----> core_ids = 0,1,3,5
2. modify the second line in generated cpu.txt.
      0,0,-1,-1  -----> 0,0,0,0

terrafin, 2019-08-06:
It worked, thanks!
I'll be waiting for uNVMe driver update.

closed 2019-08-06 by terrafin