OpenMPDK/KVSSD #37: Install error: rmmod: ERROR: Module nvme is in use

mkbahk, 2019-08-19:
**Location (Korea, USA, China, India, etc.)**
Korea

**Describe the bug**
After successful compilation I tried to install the kernel driver using ./re_insmod.sh. The following error occurred. What should I check?

**Error Messages**
root@london:~/KVSSD/PDK/driver/PCIe/kernel_driver/kernel_v4.13.15-041315-ubuntu-16_04# ./re_insmod.sh 
rmmod: ERROR: Module nvme is in use
rmmod: ERROR: Module nvme_core is in use by: nvme
insmod: ERROR: could not insert module nvme-core.ko: File exists
insmod: ERROR: could not insert module nvme.ko: File exists

root@london:~/KVSSD/PDK/driver/PCIe/kernel_driver/kernel_v4.13.15-041315-ubuntu-16_04# uname -a
Linux london 4.13.15-041315-generic #201711211030 SMP Tue Nov 21 10:31:16 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

root@london:~/KVSSD/PDK/driver/PCIe/kernel_driver/kernel_v4.13.15-041315-ubuntu-16_04# ls -al
total 1184
drwxr-xr-x  3 root root   4096 Aug 19 15:35 .
drwxr-xr-x 11 root root   4096 Aug 19 14:25 ..
-rw-r--r--  1 root root 104040 Aug 19 14:25 core.c
-rw-r--r--  1 root root 102208 Aug 19 15:35 core.o
-rw-r--r--  1 root root  50469 Aug 19 15:35 .core.o.cmd
-rw-r--r--  1 root root  26507 Aug 19 14:25 fabrics.c
-rw-r--r--  1 root root   5510 Aug 19 14:25 fabrics.h
-rw-r--r--  1 root root  79613 Aug 19 14:25 fc.c
-rw-r--r--  1 root root   1421 Aug 19 14:25 Kconfig
-rw-r--r--  1 root root  25561 Aug 19 14:25 lightnvm.c
-rw-r--r--  1 root root  20512 Aug 19 15:35 lightnvm.o
-rw-r--r--  1 root root  42726 Aug 19 15:35 .lightnvm.o.cmd
-rw-r--r--  1 root root  30337 Aug 19 14:25 linux_nvme.h
-rw-r--r--  1 root root   3445 Aug 19 14:25 linux_nvme_ioctl.h
-rw-r--r--  1 root root    408 Aug 19 14:25 Makefile
-rw-r--r--  1 root root    191 Aug 19 15:35 modules.order
-rw-r--r--  1 root root   4732 Aug 19 15:35 Module.symvers
-rw-r--r--  1 root root 105400 Aug 19 15:35 nvme-core.ko
-rw-r--r--  1 root root    447 Aug 19 15:35 .nvme-core.ko.cmd
-rw-r--r--  1 root root    542 Aug 19 15:35 nvme-core.mod.c
-rw-r--r--  1 root root   2536 Aug 19 15:35 nvme-core.mod.o
-rw-r--r--  1 root root  28470 Aug 19 15:35 .nvme-core.mod.o.cmd
-rw-r--r--  1 root root 110240 Aug 19 15:35 nvme-core.o
-rw-r--r--  1 root root    387 Aug 19 15:35 .nvme-core.o.cmd
-rw-r--r--  1 root root  10813 Aug 19 14:25 nvme.h
-rw-r--r--  1 root root  55000 Aug 19 15:35 nvme.ko
-rw-r--r--  1 root root    427 Aug 19 15:35 .nvme.ko.cmd
-rw-r--r--  1 root root   1243 Aug 19 15:35 nvme.mod.c
-rw-r--r--  1 root root   3680 Aug 19 15:35 nvme.mod.o
-rw-r--r--  1 root root  28410 Aug 19 15:35 .nvme.mod.o.cmd
-rw-r--r--  1 root root  52280 Aug 19 15:35 nvme.o
-rw-r--r--  1 root root    287 Aug 19 15:35 .nvme.o.cmd
-rw-r--r--  1 root root  71564 Aug 19 14:25 pci.c
-rw-r--r--  1 root root  54424 Aug 19 15:35 pci.o
-rw-r--r--  1 root root  44723 Aug 19 15:35 .pci.o.cmd
-rw-r--r--  1 root root  51614 Aug 19 14:25 rdma.c
-rwxr-xr-x  1 root root     96 Aug 19 14:25 re_insmod.sh
drwxr-xr-x  2 root root   4096 Aug 19 15:35 .tmp_versions
root@london:~/KVSSD/PDK/driver/PCIe/kernel_driver/kernel_v4.13.15-041315-ubuntu-16_04# 

**System environment (please complete the following information)**
 - Firmware version : ETA51KBV
 - Number of SSDs : 4EA
 - OS & Kernel version [e.g., Ubuntu 16.04 Kernel v4.9.5]:  Ubuntu 16.04 Kernel v4.13.15-041315
 - GCC version [e.g., gcc v5.0.0] : 5.4.0 20160609
[KVSSD-Kernel Driver-Install-error.txt](https://github.com/OpenMPDK/KVSSD/files/3514638/KVSSD-Kernel.Driver-Install-error.txt)

xadongfei, 2019-08-19:
Hi mkbahk,
It seems you have NVMe device is in use, you need check it first.
./re_insmod.sh needs remove the old nvme module from linux first, and then insert our nvme driver into the linux. if there is any nvme device is in use, the nvme module can not be removed.

mkbahk, 2019-08-19:
I have one m.2 intel nvme ssd and four u.2 samsung nvme kv ssds. m.2 intel nvme ssd is used for the operating system. Could this (m.2 intel ssd) cause someone to use nvme?

xadongfei, 2019-08-20:
Yes, the problem should be caused by this NVMe opertating system disk. I suggest you use a SATA or SAS disk as an operating system.

mkbahk, 2019-08-21:
Thanks you..Good supporting..

closed 2019-08-21 by mkbahk