OpenMPDK/KVSSD #38: " make rocksdb_bench " failure

mkbahk, 2019-08-21:
**Location (Korea, USA, China, India, etc.)**
Korea

**Describe the bug**

"make rocksdb_bench" failure like below.
root@london:~/KVSSD/application/kvbench/build_rxdb# make rocksdb_bench
Scanning dependencies of target rocksdb_bench
[  8%] Building CXX object CMakeFiles/rocksdb_bench.dir/bench/couch_bench.cc.o
[ 16%] Building CXX object CMakeFiles/rocksdb_bench.dir/wrappers/couch_rocksdb.cc.o
[ 25%] Building CXX object CMakeFiles/rocksdb_bench.dir/utils/thpool.cc.o
[ 33%] Building CXX object CMakeFiles/rocksdb_bench.dir/utils/avltree.cc.o
[ 33%] Building CXX object CMakeFiles/rocksdb_bench.dir/utils/stopwatch.cc.o
[ 41%] Building CXX object CMakeFiles/rocksdb_bench.dir/utils/iniparser.cc.o
[ 50%] Building CXX object CMakeFiles/rocksdb_bench.dir/utils/crc32.cc.o
[ 58%] Building CXX object CMakeFiles/rocksdb_bench.dir/utils/memleak.cc.o
[ 66%] Building CXX object CMakeFiles/rocksdb_bench.dir/utils/zipfian_random.cc.o
[ 75%] Building CXX object CMakeFiles/rocksdb_bench.dir/utils/keyloader.cc.o
[ 83%] Building CXX object CMakeFiles/rocksdb_bench.dir/utils/memory.cc.o
[ 91%] Building CXX object CMakeFiles/rocksdb_bench.dir/utils/keygen.cc.o
[100%] Linking CXX executable rocksdb_bench
/usr/bin/ld: cannot find -lssl
/usr/bin/ld: cannot find -lcrypto
/usr/bin/ld: cannot find -lz
collect2: error: ld returned 1 exit status
CMakeFiles/rocksdb_bench.dir/build.make:380: recipe for target 'rocksdb_bench' failed
make[3]: *** [rocksdb_bench] Error 1
CMakeFiles/Makefile2:252: recipe for target 'CMakeFiles/rocksdb_bench.dir/all' failed
make[2]: *** [CMakeFiles/rocksdb_bench.dir/all] Error 2
CMakeFiles/Makefile2:264: recipe for target 'CMakeFiles/rocksdb_bench.dir/rule' failed
make[1]: *** [CMakeFiles/rocksdb_bench.dir/rule] Error 2
Makefile:183: recipe for target 'rocksdb_bench' failed
make: *** [rocksdb_bench] Error 2
root@london:~/KVSSD/application/kvbench/build_rxdb# 

[kvbench-racksdb failure-moonkee--20190821.txt](https://github.com/OpenMPDK/KVSSD/files/3523331/kvbench-racksdb.failure-moonkee--20190821.txt)

mkbahk, 2019-08-21:
I got same message when build as_bench..

--------------build messages----------------------
root@london:~/KVSSD/application/kvbench/build_as# make as_bench
[  9%] Linking CXX executable as_bench
/usr/bin/ld: cannot find -lssl
/usr/bin/ld: cannot find -lcrypto
/usr/bin/ld: cannot find -lz
collect2: error: ld returned 1 exit status
CMakeFiles/as_bench.dir/build.make:380: recipe for target 'as_bench' failed
make[3]: *** [as_bench] Error 1
CMakeFiles/Makefile2:363: recipe for target 'CMakeFiles/as_bench.dir/all' failed
make[2]: *** [CMakeFiles/as_bench.dir/all] Error 2
CMakeFiles/Makefile2:375: recipe for target 'CMakeFiles/as_bench.dir/rule' failed
make[1]: *** [CMakeFiles/as_bench.dir/rule] Error 2
Makefile:222: recipe for target 'as_bench' failed
make: *** [as_bench] Error 2
root@london:~/KVSSD/application/kvbench/build_as#

jackcost6, 2019-08-21:
Hi mkbahk,
Your system lack of shared library.
In Ubuntu, you can use following comand to install shared library.
     apt-get install libssl-dev zlib1g-dev

mkbahk, 2019-08-21:
after install shared lib by "apt-get install libssl-dev zlib1g-dev",  I got success of building as_bench.
message like below:
-------------------------Start of Messages--------------------------
root@london:~/KVSSD/application/kvbench/build_as# cmake ../
-- Configuring done
-- Generating done
-- Build files have been written to: /root/KVSSD/application/kvbench/build_as
root@london:~/KVSSD/application/kvbench/build_as# make as_bench
[  9%] Linking CXX executable as_bench
[100%] Built target as_bench
root@london:~/KVSSD/application/kvbench/build_as# ll
total 652
drwxr-xr-x  3 root root   4096 Aug 22 08:46 ./
drwxr-xr-x 10 root root   4096 Aug 21 11:48 ../
-rwxr-xr-x  1 root root 578160 Aug 22 08:46 as_bench*
-rw-r--r--  1 root root   2405 Aug 21 09:18 bench_config.ini
-rw-r--r--  1 root root  11538 Aug 21 11:48 CMakeCache.txt
drwxr-xr-x 13 root root   4096 Aug 22 08:46 CMakeFiles/
-rw-r--r--  1 root root   1396 Aug 21 11:48 cmake_install.cmake
-rw-r--r--  1 root root  49178 Aug 22 08:46 Makefile
----------------------end of messages-----------------------

### but stile, for building of rocksdb_bench, same error occurs before.
-------------------start of messages---------------------------
root@london:~/KVSSD/application/kvbench/build_rxdb# cmake -DCMAKE_INCLUDE_PATH=/root/rocksdb/include/rocksdb -DCMAKE_LIBRARY_PATH=</root/rocksdb ../
-- Configuring done
-- Generating done
-- Build files have been written to: /root/KVSSD/application/kvbench/build_rxdb
root@london:~/KVSSD/application/kvbench/build_rxdb# 
root@london:~/KVSSD/application/kvbench/build_rxdb# make rocksdb_bench
[  8%] Linking CXX executable rocksdb_bench
/usr/bin/ld: cannot find -lrocksdb
collect2: error: ld returned 1 exit status
CMakeFiles/rocksdb_bench.dir/build.make:380: recipe for target 'rocksdb_bench' failed
make[3]: *** [rocksdb_bench] Error 1
CMakeFiles/Makefile2:252: recipe for target 'CMakeFiles/rocksdb_bench.dir/all' failed
make[2]: *** [CMakeFiles/rocksdb_bench.dir/all] Error 2
CMakeFiles/Makefile2:264: recipe for target 'CMakeFiles/rocksdb_bench.dir/rule' failed
make[1]: *** [CMakeFiles/rocksdb_bench.dir/rule] Error 2
Makefile:183: recipe for target 'rocksdb_bench' failed
make: *** [rocksdb_bench] Error 2
root@london:~/KVSSD/application/kvbench/build_rxdb#
--------------------end of messages----------------------
 is this correct?
DCMAKE_INCLUDE_PATH=/root/rocksdb/include/rocksdb -DCMAKE_LIBRARY_PATH=</root/rocksdb ../

jackcost6, 2019-08-22:
Hi mkbahk, 
From error message, there is no NewLRUCache in rocksdb library.
I guess rocksdb you used is not supported by kvbench.
kvbench supports rocksdb tested version: v5.0.2, v5.6.1 and v5.8, please make sure this.

mkbahk, 2019-08-22:
I see. will check again..

closed 2019-08-22 by mkbahk