OpenMPDK/KVSSD #39: About bitmask and bit pattern when using iterator

terrafin, 2019-08-30:
According to specification document, KVSSD iterator operation returns sets of keys that satisfies
'(bitmask & key) == bit_pattern'.
However, while bitmask and bit_pattern's type is unsigned int, key's type is an array of char.
Then, does KVSSD iterator only cares about 1st to 4th letter of key, when doing '&' operation?

xadongfei, 2019-09-02:
Yes, this software is based on Samsung PM983, it currently supports only up to four bytes of keys for the iterator condition.

terrafin, 2019-09-02:
Then, should I care about bit representation(little-endian and big-endian)?
For example, if key is "abcd" and bitmask is 0x10000000, which one is right representation for bitmask & key : 0x10000000 & 0x64636261, or 0x10000000 & 0x61626364?

xadongfei, 2019-09-02:
Both the host soft and device shall use little endian memory and transport format.  For your example, you should use 0x10000000 & 0x64636261.

terrafin, 2019-09-02:
Got it. Thanks a lot!

closed 2019-09-02 by terrafin