OpenMPDK/KVSSD #42: KVSSD utilization in sample_test_async

mattJS17, 2019-09-24:
**Location (Korea, USA, China, India, etc.)**
Korea

**Describe the bug**
In running sample_test_async,  KVSSD's capacity is printed well. However, KVSSD's utilization is always printed as 0.

**To Reproduce**
Steps to reproduce the behavior:
1.  sudo ./sample_code_async -d /dev/nvme1n1 -n 524288 -q 64 -o 1 -k 16 -v 1048576
sudo 

**Expected behavior**
The total size written is 512GB so in sample_code_async, KVSSD's utilization should be printed as 143.

**Screenshots**
[image]

**System environment (please complete the following information)**
 - Firmware version : ETA51KBE
 - Number of SSDs : 1
 - OS & Kernel version :, Ubuntu 16.04 Kernel v4.9.5  
 - GCC version [e.g., gcc v5.0.0] : 5.4.0
 - kvbench version if kvbench runs [e.g., v0.6.0]: 
 - KV API version [e.g., v0.6.0] : the most recent one provided in Github
 - User driver version : 
 - Driver [Kernel or user driver or emulator] : kernel driver

**Workload**
 - 524288
 - sample_code_async
 - key size : 16
 - value size : 1048576
 - operation option if available [e.g., sync or async mode] : async

**Additional context**
Add any other context about the problem here.

hao86yan, 2019-09-25:
The implementation of the utilization in the firmware and KV API. Please use the latest firmware to test (ETA51KBV).

mattJS17, 2019-11-13:
It worked with the latest firmware.

Thanks

closed 2019-11-13 by mattJS17