OpenMPDK/KVSSD #45: [KVSSD KDD] Unit test and SNIA API test fail

manojps, 2019-11-04:
I have updated the firmware successfully (getting the correct firmware version in $sudo nvme list). Installed KDD on Ubuntu 18.04 LTS on  kernel version 4.15.0-66-generic. The unit tests for kernel driver failed. 
$ sudo ./kv_store_16 -d /dev/nvme0n1 -k student0 -value KJKJKJ
[sudo] password for msaha002: 
[dump issued cmd opcode (81)]
	 opcode(81)
	 flags(00)
	 rsvd1(0000)
	 nsid(00000001)
	 cdw2(00000000)
	 cdw3(00000000)
	 rsvd2(00000000)
	 cdw5(00000000)
	 data_addr(0x55a7e9742260)
	 data_length(00001000)
	 key_length(00000010)
	 cdw10(00000000)
	 cdw11(00000000)
	 cdw12(64757473)
	 cdw13(30746e65)
	 cdw14(00000000)
	 cdw15(00000000)
	 timeout_ms(00000000)
	 result(00000000)

opcode(81) error(-1) and cmd.result(0) cmd.status(0).
fail to store for /dev/nvme0n1

When I try API test code for SNIA, I get the following error.
$ sudo ./sample_code_async -d /dev/nvme0n1 -n 1000 -q 64 -o 1 -k 16 -v 4096
ENTER: open
fail to set_aioctx
EXIT : open
KV device is closed: fd 3
kv_initialize_device failed 0x20 - KVS_ERR_UNCORRECTIBLE
Device open failed 0x13

What could be the issue?

ramziammari, 2019-11-04:
This is usually happens when KV Kernel driver is not installed properly. please re-run re_insmod.sh in
KVSSD/PDK/driver/PCIe/kernel_driver/kernel_{kernel_version}/re_insmod.sh and try again

manojps, 2019-11-04:
Thank you. Reinstalled the kernel driver. It is working now.

ramziammari, 2019-11-04:
glad it worked for you. closing the issue . thanks

closed 2019-11-04 by ramziammari