OpenMPDK/KVSSD #46: ASYNC shows lower insert performance than SYNC when the value size is random

mattJS17, 2019-11-12:
**Location (Korea, USA, China, India, etc.)**
Korea

**Describe the bug**
When the value size is random, async shows lower insert performance than sync

**To Reproduce**
Steps to reproduce the behavior:
1.  format KV SSD
2. set kv_bench config with body_length distribution as uniform, upper_bound as 2MB, and lower bound as 512KB
3. run kv_bench with sync option and async option

**Expected behavior**
When the value size is fixed, async shows higher performance than sync. I Expected the similar behavior when the value size was random.

**Screenshots**
[image]

**System environment (please complete the following information)**
 - Firmware version : ETA51KBV
 - Number of SSDs : 1
 - OS & Kernel version [e.g., Ubuntu 16.04 Kernel v4.9.5]:  Ubuntu 16.04 Kernel v4.9.5
 - GCC version [e.g., gcc v5.0.0] : 5.4.0
 - kvbench version if kvbench runs [e.g., v0.6.0]: the most recent one provided in Github
 - KV API version [e.g., v0.6.0]: the most recent one provided in Github
 - User driver version :
 - Driver [Kernel or user driver or emulator] : kernel driver

**Workload**
 - 10000
 - sequential insert using kv_bench
 - key size :  16B
 - value size : uniform(512KB ~ 2MB)
 - operation option if available [e.g., sync or async mode] : sync & async

ramziammari, 2019-11-12:
Hi, 
Can I have a copy of your "bench_config.ini" since I am not able to re-produce this issue. 

Thanks

mattJS17, 2019-11-13:
Hi,
I attached the async config file. When I ran with sync option, I changed both [kvs] -write mode and [operation] -write type to sync.
[bench_config.txt](https://github.com/OpenMPDK/KVSSD/files/3839430/bench_config.txt)

* I tested the sequential insert performance using population. So the write ratio or read ratio doen't matter
*  I did change the value_pool_unit
* I've set both [kvs] -write mode and [operation] -write type to sync or async.

Thanks

closed 2019-11-22 by mattJS17