OpenMPDK/KVSSD #47: kernel support update needed: kernel-devel-3.10.0-1062.9.1.el7

jammingyu9, 2020-01-06:
**Location (Korea, USA, China, India, etc.)**
South Korea

**Describe the bug**
kernel-devel-3.10.0-1062.9.1.el7 (which was published in Dec. 9) deprecates the function below
- [nvme] blk-mq: remove blk_mq_complete_request_sync (Ming Lei) [1763624 1730922]
so that the driver of kernel_v3.10.0-1062-centos-7_7 won't be compiled.

**To Reproduce**
Steps to reproduce the behavior:
1.  yum update
2. yum install kernel-devel
3. compile the KVSSD driver for kernel_v3.10.0-1062-centos-7_7

**Screenshots**
[image]

**System environment (please complete the following information)**
 - Firmware version : -
 - Number of SSDs : 2
 - OS & Kernel version [e.g., Ubuntu 16.04 Kernel v4.9.5]: Centos kernel-3.10.0-1062.9.1.el7 
 - GCC version : gcc v7.3.1
 - User driver version :
 - Driver [Kernel or user driver or emulator] :

yuanyinggao, 2020-01-07:
@Jammingyu9, this module is developed base on kernel-3.10.0-1062.el7. You can rollback kernel to 3.10.0-1062.el7 and try again. If you have reason other than have a try at KVSSD, please let us know. We may consider supporting the updated version. Thanks.

jammingyu9, 2020-04-07:
Well, this issue is a request for an update. (I've already tried it out on previous kernels)
I hope KVSSD will provide an update for new kernels for its availability.