OpenMPDK/KVSSD #52: Rocksdb-SPDK compilation failure

DanlinJia, 2020-05-02:
**Location (Korea, USA, China, India, etc.)**
USA

**Describe the bug**
Rocksdb-SPDK compilation failed due to the libspdk_event_rpc library that cannot be found for SPDK with the version above 17 (18, 19).

**To Reproduce**
Steps to reproduce the behavior:
1.  go to /KVSSD/application/kvbench/spdk/spdk/ to build spdk. The error occurs that head files cannot be found.
2. Download and build spdk-18.04 from GitHub to replace /KVSSD/application/kvbench/spdk/spdk/.
3. go to /KVSSD/application/kvbench/spdk/rocksdb/ to build rocksdb. Succeed.
4. copy necessary libraries for spdk, dpdk and rocksdb to /usr/local/lib/
5. cd kvbench
    mkdir build_spdk && cd build_spdk
    cmake ../
    make blobfs_rocksdb_bench
6. error: -lspdk_event_rpc not found.
7. Download and compile spdk-17.10 from Github to replace /KVSSD/application/kvbench/spdk/spdk/, copy libspdk_event_rpc.a to /usr/local/lib. 
8. ''No reference'' error happens.

**Expected behavior**
compile Rocksdb-SPDK successfully.

**System environment (please complete the following information)**
 - Firmware version : ETA51KCA_20191021_ENC.bin
 - Number of SSDs : 1
 - OS & Kernel version [e.g., Ubuntu 16.04 Kernel v4.9.5]: Ubuntu 16.04 Kernel v4.13.15
 - GCC version [e.g., gcc v5.0.0] :5.4.0
 - User driver version : spdk-18.04
 - Driver [Kernel or user driver or emulator] : User Driver 

**Additional context**
1. Benchmark test on kernel device driver succeeds for kvbench and rocksdb.
2. The sample code test for the user device driver succeeds via uNVMe.

hao86yan, 2020-05-05:
We set up the same environment in our lab and compiled the SPDK successfully. Please comfirm that if all dependencies were installed. 
For ubuntu:  sudo apt-get install libsnappy-dev libev-dev libbz2-dev liblz4-dev libzstd-dev libjemalloc-dev libnuma-dev libgflags-dev libssl-dev libz-dev
For CentOS:  sudo yum install snappy-devel libev-devel bzip2-devel lz4-devel gflags-devel jemalloc-devel openssl-devel zlib-devel

If you still have the compilation issue, please provide more detailed error information ?

DanlinJia, 2020-05-09:
Issue solved by changing the privilege of files to executable, under folders ''scripts/'' and "dpdk/buildtools/".

closed 2020-05-09 by DanlinJia

manojps, 2020-05-12:
Changing permissions did not work for me. Trying to figure out what's different in my setup..