OpenMPDK/KVSSD #53: SPDK Device open failed 0x3

manojps, 2020-05-07:
I was using KDD in the machine before. When I moved to SPDK, I get the following error. Could please advise on this issue.

$ sudo ./sample_code_async -d 0000:06:00.0 -n 100 -q 64 -o 1 -k 16 -v 4096
EAL: Detected 32 lcore(s)
EAL: Detected 2 NUMA nodes
EAL: Auto-detected process type: PRIMARY
EAL: Multi-process socket /var/run/dpdk/spdk4251/mp_socket
EAL: No free hugepages reported in hugepages-1048576kB
EAL: Probing VFIO support...
kv_driver.c:: 219:: _kv_env_init:: mem_size_mb: 1024 shm_id: 4251

kv_driver.c:: 222:: _kv_env_init:: Initialized the KV API Environment
kv_driver.c:: 243:: _kv_env_init:: Done

init udd
kv_driver.c:: 372:: kv_nvme_init:: Cannot Use the Requested Device 0000:06:00.0
Device open failed 0x3

$ sudo nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     S4DRNY0MXXXXXX       SAMSUNG MZQLB3T8HALS-000KV               1           1.46  GB /   3.84  TB    512   B +  0 B   ETA51KCA
/dev/nvme1n1     S4DRNY0MXXXXXX       SAMSUNG MZQLB3T8HALS-000KV               1           0.00   B /   3.84  TB    512   B +  0 B   EDA53W0Q
===

hao86yan, 2020-05-07:
We suspect that you might skip some essential steps during setting up the SPDK environment. Please refer to KVSSD_QUICK_START_GUIDE_1.1.0.pdf and follow the steps in section 3.1.5 to set up the running environment of SPDK.

DanlinJia, 2020-05-09:
While running the sample code, I had same issue and solved by replacing ''-d 0000:06:00.0'' with ''-d 0000:17:00.0''.

manojps, 2020-05-12:
For me the device name was different. The device name is reported when we load the SPDK driver using KVSSD/PDK/core/tools/setup.sh or can also be found using lspci |grep -i nvme command.

closed 2020-05-12 by manojps