OpenMPDK/KVSSD #55: [kvbench] Question about KVBench configuartion options

manojps, 2020-05-10:
1. What is the difference between [kvs] write_mode and [operation] write_type?
2. What are the supported options for [kvs] store_option, other than post?
3. What is the difference between [threads] readers and reader_ops, or writer and writer_ops?
4. What is the purpose of [population] pop_first?
5. We have thread configuration options under multiple section in the config file - [kvs]aiothreads_per_device, [population] nthreads and an entire [threads] section. What are the differences between these? 
6. If we don't want I/Os to be cached for direct KV access, which options should I chenge in the bench_config.ini file and which values should be set? I have noticed mismatch between both KVS_ops.txt and smartctl logs in terms of I?o count and data read. However, such discrepancies doesn't exist when using sample codes provided in the repo for I/Os.

hao86yan, 2020-05-11:
Pelease see the reply in line. 
1. What is the difference between [kvs] write_mode and [operation] write_type?
   "[kvs] write_mode" indicates which type of APIs will be invoked. KVS API has 2 types of API: aysnchronous APIs and synchronous APIs. 
   "[operation] write_type" only works for RocksDB, not for KV SSD.  It indicates the data is wrote to disk synchronously or firsly to memory and then to disk asynchronously. 

2. What are the supported options for [kvs] store_option, other than post?
  For KV SSD, only post is supported.

3. What is the difference between [threads] readers and reader_ops, or writer and writer_ops?
   Readers and writers represent the number of threads for read and write operation
   reader_opts and writer_ops were not used so far.

4. What is the purpose of [population] pop_first?
    pop_first indicates if it's is needed to pre-write data to SSD before test.

5. We have thread configuration options under multiple section in the config file - 
    [kvs]aiothreads_per_device is unused.
    [population] nthreads means the number of pre-write threads.
    [threads] means the number of test threads.

6. If we don't want I/Os to be cached for direct KV access, which options should I chenge in the bench_config.ini file and which values should be set? I have noticed mismatch between both KVS_ops.txt and smartctl logs in terms of I?o count and data read. However, such discrepancies doesn't exist when using sample codes provided in the repo for I/Os.  Your documentation says "allocator = posix # 'posix' or 'numa' for block device; 'spdk' for KV SSD only". What should be allocator for KDD?
     For KVAPI testing, there is no cache functionality in kvbench

manojps, 2020-05-21:
7. Only [kvs] with_iterator = alone with [population] seq_fill = false returns throughput for iterator. What is the difference between "with_iterator = alone" and "with_iterator = true"? And how should we interpret the throughput results of the benchmark with these two options?

8. How to open multiple iterators using KVBench?

9. How to set the bitmask and bit_pattern for iterators in KVBench? 

10. During insertion phase, we can control sequential inserts by modifying [population] seq_fill settings. Is there a way to do strictly sequential read, write and inserts during benchmark phase using KV direct access in KVBench, instead of using a KV store like RocksDB?

hao86yan, 2020-05-27:
For question 7,8, and 9, we removed the test support of iterator in kv_bench, because kv_bench is originally developed for RocksDB and aerospike, so it doesn't supported iterator well. If you want to test iterator, we recommend to use our sample code.

10. During insertion phase, we can control sequential inserts by modifying [population] seq_fill settings. Is there a way to do strictly sequential read, write and inserts during benchmark phase using KV direct access in KVBench, instead of using a KV store like RocksDB?
  Please refer to the section 3 of the readme file in kvbench folder to compile kv_bench running on KV SSD directily.  The configuration of seq_fill also takes effect in direct KV SSD access.

11. When we're using KDD, should we set the [system] allocator = kdd?
  There is no need to setting the allocator to KDD when you use KDD.

12. What's the difference between key_existing = true and = false?
  When key_existing is set to true, kv_bench guarantees the keys already existed in KV SSD when doing read, delete, and update operations.

13. What does [kvs] mem_size_mb set?
  This configuration item has not been used.