OpenMPDK/KVSSD #56: No SPDK allocator available, please use Posix or Numa

manojps, 2020-06-20:
I am getting the following error when running rocksDB on SPDK.

```
$ sudo LD_LIBRARY_PATH=/KVSSD/application/kvbench/build_spdk/ ./blobfs_rocksdb_bench -f bench_config.ini
Using "bench_config.ini" config file
No SPDK allocator available, please use Posix or Numa
```
Initialization of blobfs:
```
# /KVSSD/application/kvbench/spdk/spdk/test/lib/blobfs/mkfs/mkfs /tmp/rocksdb.conf Nvme0n1
Starting DPDK 17.05.0 initialization...
[ DPDK EAL parameters: spdk_mkfs -c 0x3 -m 1024 --file-prefix=spdk_pid3668 ]
EAL: Detected 32 lcore(s)
EAL: No free hugepages reported in hugepages-1048576kB
EAL: Probing VFIO support...
Total cores available: 2
Occupied cpu socket mask is 0x1
reactor.c: 362:_spdk_reactor_run: *NOTICE*: Reactor started on core 1 on socket 0
reactor.c: 362:_spdk_reactor_run: *NOTICE*: Reactor started on core 0 on socket 0
EAL: PCI device 0000:87:00.0 on NUMA socket 1
EAL:   probe driver: 144d:a808 spdk_nvme
EAL: PCI device 0000:af:00.0 on NUMA socket 1
EAL:   probe driver: 8086:f1a6 spdk_nvme
EAL: PCI device 0000:b0:00.0 on NUMA socket 1
EAL:   probe driver: 8086:f1a6 spdk_nvme
Initializing filesystem on bdev Nvme0n1...done.
```
rocksbd.conf file settings:
```
[Nvme]
  TransportID "trtype:PCIe traddr:0000:87:00.0" Nvme0
```
bench_config.ini settings:
```
[system]
allocator = spdk
key_pool_size = 128
key_pool_unit = 16
key_pool_alignment = 4096
value_pool_size = 128
value_pool_unit = 4096
value_pool_alignment = 4096
device_path = 0000:87:00.0

[kvs]
device_path = 0000:87:00.0 
emul_configfile = /KVSSD/PDK/core/kvssd_emul.conf 
store_option = post
queue_depth = 64
aiothreads_per_device = 1
core_ids = 0,2,4
cq_thread_ids = 2,4,6
mem_size_mb = 1024
write_mode = sync
with_iterator = false #alone # true/
iterator_mode = key
```

hao86yan, 2020-06-22:
SPDK alloctor only take effects in testing KV API. Please set the allocator to Posix or Numa when you test RocksDB

manojps, 2020-06-22:
Does that mean, RocksDB on SPDK is running on a block interface?

hao86yan, 2020-06-22:
The allocator is only used for the memory allocatio and release of kv_bench. It doesn't affect how RocksDB running o SPDK. It only used in the code of kv_bench.