OpenMPDK/KVSSD #70: Hardware compatibility

lrowe, 2023-11-02:
Location: USA

Which hardware models are compatible with KV-SSD? Are consumer drives supported?

I'm a software engineer looking wondering if I can develop locally without needing data centre server hardware. I have tried contacting sales support for an answer but have now been redirected four times without anyone knowing anything.

Neurone, 2023-11-15:
Same question here; I am still searching the Internet for an answer; I cannot find any vendor that specifies the command sets it supports. For sure, I can say I just recently discovered that[ my Samsung 990 Pro does not support it](https://github.com/OpenMPDK/xNVMe/issues/134#issuecomment-1267116997), even if it's an NVMe 2.0 SSD drive. It does not even support [querying for the supported Command Sets](https://github.com/OpenMPDK/xNVMe/issues/379), just to make things more _spicy_, but that's another story :)

I'm trying to understand now if the [Crucial T700](https://www.crucial.com/products/ssd/crucial-t700-ssd) supports KVCS, before buying it this time, if possible.

Neurone, 2023-11-20:
Yeah, you are probably right; it is very likely they will try at least to leverage these features in marketing; as we have seen, it's still very rare to find a HW supporting KVCS, and if even a single SSD exists already, I was unable to prove it yet.

As an update for the list, I can confirm **Crucial T700 does not support KV Command Set** (I got an answer from their support team directly). Because T700 uses the [Phison E26](https://www.phison.com/en/gen5-e26-ssd-partner-ecosystem) controller, I would say we can consider any other SSDs based on that controller as a non-supporting KVCS. I asked Phison if they have or plan to release a controller supporting KVCS; I'm still waiting for an answer.